joeyoung/arduino_keypads #5: Lack of license

MiLeG, 2020-04-13:
Greetings,
I'd love to use your code as a part of my own project, but there isn't a defined license on this repository.
According to this [page](https://choosealicense.com/no-permission/), it's best to ask for a license first than to use unlicensed code in the first place.
Please, consider adding a license to your code, it would help me out a lot!

Thank you for your time!

asokolsky, 2020-04-13:
Sure.
I’ll add mit license.

asokolsky, 2020-04-13:
I responded to this e-mail by mistake.  My apologies to the owner of the repository.
Cheers!

ldrrp, 2020-04-13:
MIT please :)

joeyoung, 2020-04-14:
Thanks for bringing this to my attention. I am presently in the process of revising the libraries to fix a problem that arose with MKR and ESP32 boards. As these revisions roll out, I will include a license text with each library. As these libraries are only useful as an addition to the Keypad library, I'll adopt the same license used there: GNU General Public License 3.0.

closed 2020-04-14 by joeyoung