joeyoung/arduino_keypads #7: Keypad_I2C doesn't register key press

adrianbn, 2020-06-22:
Hi,

Thanks for this library! I downloaded the Keypad_I2C library and the CustomKeypad example to test my 4x4 membrane keypad with my I2C expander board. The board uses PF8574T (https://smile.amazon.com/gp/product/B07XD2K4GH). I manage to get an I2C port scanner to find the PF8574T at address 0x20 (all jumpers off), but when I load the CustomKeypad example there are no key presses detected. Worth mentioning that the only change I made to the example code was to replace 0x21 with 0x20 as that is the address my module lives on.

I have wired the keypad directly to the arduino and that works fine. I've also tried a second PF8574T chip (same breakout board model, I just got 2) to rule out chip failure. 

I'm not quite sure how the Keypad library and this library are wired together, but I was able to trace the getKey -> getKeys -> scanKeys -> readPin call into this library and added some debug to readPin and writePin. The output below has been recorded while repeatedly pressing key 1.

```
Pin 1 has value: 2 -> 1
Pin 0 has value: 1 -> 1
pin_write on pin 5 with level 1. Calling port_write with 65535
pin_write on pin 4 with level 0. Calling port_write with 65519
Pin 3 has value: 8 -> 1
Pin 2 has value: 4 -> 1
Pin 1 has value: 2 -> 1
Pin 0 has value: 1 -> 1
pin_write on pin 4 with level 1. Calling port_write with 65535
pin_write on pin 7 with level 0. Calling port_write with 65407
Pin 3 has value: 8 -> 1
Pin 2 has value: 4 -> 1
Pin 1 has value: 2 -> 1
Pin 0 has value: 1 -> 1
pin_write on pin 7 with level 1. Calling port_write with 65535
pin_write on pin 6 with level 0. Calling port_write with 65471
Pin 3 has value: 8 -> 1
Pin 2 has value: 4 -> 1
Pin 1 has value: 2 -> 1
Pin 0 has value: 1 -> 1
pin_write on pin 6 with level 1. Calling port_write with 65535
pin_write on pin 5 with level 0. Calling port_write with 65503
```

Any help debugging this issue would be highly appreciated!

joeyoung, 2020-06-22:
Thanks for noticing this problem. I think that I've left a testing version of DynamicKeypad in the Keypad_I2C library. The Setup for all of the example sketches requires Wire.begin( ), which was commented out in the library example. I've just uploaded a corrected version. Please verify that it now works properly and post a closing comment.

adrianbn, 2020-06-23:
That did it, thank you!

closed 2020-06-23 by adrianbn