joeyoung/arduino_keypads #8: Multiple expansion port chips

snotfod, 2020-06-29:
Hi, thank you for your work and updates. I really like it and use it on several projects.
I was wondering if it would be too much to ask for an example or help using 2 chips, so we can break the 8x8 barrier and make a full keyboard? 
Maybe one chip for columns and the other for rows or use pins 16 to 31 for the second chip.
I've tried to make it myself with 2x mcp23017 but so far without success.

joeyoung, 2020-06-29:
The example HelloKpd64 in the Keypad_MC17 library shows one way for
using (up to) 4 MCP23017s.

The concern becomes the length of time it takes to scan all the keys.
The time can be reduced by decreasing the debounce time, setting the I2C
bus speed to 400000, or using a faster processor. The example has a
sketchy few examples of different timings.

snotfod, 2020-06-29:
Thank you for taking the time to respond.
I use an esp32.
I have an older keyboard that is 8x16.
After looking at your example, I think it must be possible to convert it to 2x 8x8 by sharing the rows between the chips with diodes.
Thank you.

By the way, I saw in your README.md that all MCP chips have a 0 too much.

joeyoung, 2020-06-29:
You can probably get away without diodes. Keypad leaves all pins as inputs after scanning, so almost any configuration can work. For example, you could connect the 8 side to the column pins on both MCP23017s, half the 16 side to one MCP23017 row pins, and the other half to the other row pins. The column pins on both IC would be driven with the first chip scan (and vice-versa for second) but since the second one's column lines are input there would be no harm. Keypad drives the column lines and reads the row lines.

joeyoung, 2020-06-29:
Thanks for the correction of the MCP part numbers.

closed 2020-06-29 by joeyoung

snotfod, 2020-06-30:
Thanks you, I did as you suggested and it works great.