joeyoung/arduino_keypads #10: Help: Not working with PCF8575

rahmanshaber, 2020-07-22:
This is my code and schematics , I am getting "O" "E" when connecting A1 and P00, 
A littel help will be apprecited. i tested all the pins with a led with this lib and all are working as expected.
https://github.com/skywodd/pcf8574_arduino_library/blob/master/PCF8575/examples/ButtonBlink/ButtonBlink.ino

[image]

```

#include <Keypad_I2C.h>
#include <Keypad.h>        // GDY120705
#include <Wire.h>

#define I2CADDR 0x20

const byte ROWS = 5; //four rows
const byte COLS = 11; //three columns

char hexaKeys[ROWS][COLS] = {
  {'P', 'O', 'E', 'E', 'E', 'E', 'E', 'E', 'E', '\\', 'i'},
  {'1', '2', '3', '4', '5', '6', '7', '8', '9', '0', '0'},
  {'q', 'w', 'e', 'r', 't', 'y', 'u', 'i', 'o', 'p', '0' },
  {'0', 'a', 's', 'd', 'f', 'g', 'h', 'j', 'k', 'l', '0'  },
  {'0', '0',  'z', 'x', 'c', 'v', 'b', 'n', 'm', 'i', '0'  }
};
byte rowPins[ROWS] = {13, 17, 2, 4, 5}; //connect to the row pinouts of the keypad
byte colPins[COLS] = {6, 11, 10, 7, 12, 14, 15, 16, 3, 1, 0}; //connect to the column pinouts of the keypad

//initialize an instance of class NewKeypad
Keypad_I2C customKeypad( makeKeymap(hexaKeys), rowPins, colPins, ROWS, COLS, I2CADDR, PCF8575); 

void setup(){
  Wire.begin( );                // GDY200622
  customKeypad.begin( );        // GDY120705
  Serial.begin(9600);
}
  
void loop(){
  char customKey = customKeypad.getKey();
  
    Serial.println('u');
    delay(500);
  if (customKey != NO_KEY){
    Serial.println(customKey);
  }
}

joeyoung, 2020-07-22:
The keyboard connection would join a row to a column when a key is pressed. So to get an 'O', you would connect the 2nd column to the first row--pin p17 to p06 of the PCF8575. I don't understand what your connection to A1 (ground?) is meant to do. I suppose that A,B,C,D, E are the labels on the keyboard's row pins, and the number labels are on the columns, based on the numbering of the rowPins[ ] and colPins[ ] definitions.

rahmanshaber, 2020-07-23:
[image]

A1,2,3 are connected to ground in my circuit as i followed the datasheet. and yes A B C D E are the rows.

[image]

joeyoung, 2020-07-23:
OK, but your original comment said that you connect A1 and P00, so I do not understand what you are asking. Is A1 the name of a keyboard key? The diagram looks like it starts with A2, connecting column 2 to row A which should give character 'P'. Does that happen?

rahmanshaber, 2020-07-23:
No, as no other pins giving me output then connecting the P00 to GND/A1 of the PCF ic. 
A1 is from the PCF8575.
don't mind those A1~A10, those are just added by the eagle, i am talking about the PCF8575 A1,2,3 pins.

Is the declaration of the row and column pins are correct? as in the datasheet there is no 8,9,10 pin, they started the portB with 11, not 8
[image]

rahmanshaber, 2020-07-23:
Also how the keypad lib knows that those pins user  added in the constructor are from the PCF8575? and not from the ardinuo? 
i think there should be a way to diffrenciet between PCF pins and the ardinuo pins, i think adding p before the PFC pins will be better.

joeyoung, 2020-07-23:
The pin numbering when using the Keypad_I2C library are for the PCF8574 or the PCF8575. The library does not know about the arduino pins and cannot refer to them. The numbers 0..7 correspond to P00..P07; numbers 8..15 to P10..P17. Specifying the PCF8575 in the constructor determines that all the accesses will be for a 16-bit port.
For the Keypad library functions, the only connections to the port pins are made by the keyboard and they only connect a row pin to a column pin. There is no connection between the P00..P17 pins and ground or to any other pin on the PCF8575.

rahmanshaber, 2020-07-23:
Okay, so what might be the issue for my one, Can you suggest some solution? 

Also can you try out the code by your self? also did did tried out the PFC8575 physically?

joeyoung, 2020-07-23:
I don't know what the problem might be. You said at the start that you were getting 'O' and 'E'; is that when you press a keyboard key?
I have tried all the libraries and examples at one time or another. I don't have a PCF8575 wired up on a breadboard at the moment. I can hook it up tomorrow and check it out again.

rahmanshaber, 2020-07-23:
Yes, i was getting the O and E when connecting the A1/ GND to P00.

That will be great.

joeyoung, 2020-07-23:
OK, I have wired up a PCF8575, and loaded in your sketch as above. It seems to work fine, although I do not have a 5X11 keypad. I am testing by just jumpering a few of the key positions. For example if I connect port pin 13 (first row) to 6, 11, 10, ... I get the first row characters mixed in with the steady stream of 'u'.
I do notice that one of your column pin specifications is 16 which doesn't exist...only 0..15, maybe that should be 8 as that's the only one missing. This does not stop the other column pin scanning though, so other than missing one column of keys your program should work.
The delay(500) in your loop will cause fairly slow response, because while in the delay, the keyboard scanning is stopped.
I assume that your assigning the rows and columns as shown is to make connections between the keyboard and the PCF8575 simpler, but maybe there is a wiring error that I can't see.

joeyoung, 2020-07-23:
I see that I've been ambiguous with the signal names. The numbers in the note above are the logical bit numbers as specified in the constructor lists. Logic pin 13 is the IC pin 18 named P15.

rahmanshaber, 2020-07-24:
okay, maybe the delay causing not to get all the output, i will try wth out it.

and can you plese fix/change this

- change the pin numbers in the contractor to P1-p7 and P10-P17, that way it will match the datasheet.
- the code should not build/give an error if the pin assigning is wrong, like what i did with pin 16.
- also it should not build if user did not added the chip name in the contractor, it took me some time to figure it out that i did not added that and that is causing some issue.
- also add readme in every lib folder, i did not find the docs folder easily,

joeyoung, 2020-07-24:
If you want to use the port pin names, add this to your sketch (maybe use another tab):
[code]
// map logical pin numbers to PCF8575 datasheet port pin names
//
#define P00 0
#define P01 1
#define P02 2
#define P03 3
#define P04 4
#define P05 5
#define P06 6
#define P07 7
#define P10 8
#define P11 9
#define P12 10
#define P13 11
#define P14 12
#define P15 13
#define P16 14
#define P17 15
[/code]

There is not a way to detect on construction whether an invalid pin name is used. A run-time error might be possible, but the arduino environment is bad at reporting runtime errors.

The library defaults to using the PCF8574 when no name is specified for a convenience and historical reasons since the original development and most common use is with the PCF8574. It is not detectable for an error message if you leave out the specification for this reason.

If I add a readme to every folder, it becomes a maintenance problem to ensure that all the readme's are up to date if something changes. I will consider adding a 'please see the main readme', but it is standard library format to place library-use information in the 'documentation' folder within the library.

rahmanshaber, 2020-08-01:
Okay, i got it working, Thanks Alot for the library.

If you are intersedted check out my project where i will use this library.

https://mutantc.gitlab.io/

closed 2020-08-01 by rahmanshaber