joeyoung/arduino_keypads #11: Multiple Keypad_I2C Objects

Maes2ro, 2020-08-25:
I would like to be able to use 4 keypads simultaneously, and have built a circuit with four PCF8574 IC's, each with unique addresses.
To test this, I have created a modified version of your HelloKeypad_I2C example, in which I declare four separate Keypad_I2C objects. These declarations are slightly simplified, in that I removed the jwire reference. This is entirely because I don't understand it, and it's not in all your other examples, although of course it would be better if I did understand!
Havng declared the four objects, I then .begin them all. It is interesting to note that sometimes the .print of the pinState shows FF, and other times FFFFFFFF. In the latter case, nothing works at all.
What I have found is that (provided pinState is FF), only the first getKey statement in the sequence works. They all can work provided that I comment out the others.
In summary then I have three questions, in order of priority:
What am I doing wrong with the multiple getKey calls?
What might be causing the FF vs FFFFFFFF pinState report?
What is that jwire pointer doing? (this is just for my own interest and [education)

I would greatly appreciate any help and guidance you can give me on this.

Full Code Below

/* @file HelloKeypad.ino

   Revised to use I2C i/o Feb 26/12 - G. D. Young
   Re-check with Keypad 3.0, arduino 1.0.1 Jul 29/12
   Add port-type parameter to constructor for Paul
   Williamson's modification of the Keypad_I2C library Sept 2/13

   April 6/2020 - Keypad_I2C allows optional WireX parameter. Note
   that if WireX parameter is spec'd, then the otherwise optional
   'width' parameter is needed. OK to omit both options when using
   8 bit port chips on Wire

|| @version 1.0
|| @author Alexander Brevig
|| @contact alexanderbrevig@gmail.com
||
|| @description
|| | Demonstrates the simplest use of the matrix Keypad library.
|| #
*/
#include <Keypad_I2C.h>
#include <Keypad.h>
#include <Wire.h>

/*-----( Declare Constants )-----*/
#define I2CADDR_0 0x20
#define I2CADDR_1 0x21
#define I2CADDR_2 0x22
#define I2CADDR_3 0x23

/*-----( Declare Variables )-----*/
const byte ROWS = 4; //four rows
const byte COLS = 4; //four columns
char keys[ROWS][COLS] = {
  {'1','2','3','A'},
  {'4','5','6','B'},
  {'7','8','9','C'},
  {'*','0','#','D'}
};

byte rowPins[ROWS] = {4, 5, 6, 7}; //connect to the row pinouts of the keypad
byte colPins[COLS] = {0, 1, 2, 3}; //connect to the column pinouts of the keypad

/*-----( Declare objects )-----*/
Keypad_I2C kpd0( makeKeymap(keys), rowPins, colPins, ROWS, COLS, I2CADDR_0, 1 );
Keypad_I2C kpd1( makeKeymap(keys), rowPins, colPins, ROWS, COLS, I2CADDR_1, 1 );
Keypad_I2C kpd2( makeKeymap(keys), rowPins, colPins, ROWS, COLS, I2CADDR_2, 1 );
Keypad_I2C kpd3( makeKeymap(keys), rowPins, colPins, ROWS, COLS, I2CADDR_3, 1 );

void setup(){
  Serial.begin(9600);
  while( !Serial ){ /*wait*/ }
/*  Start I2C  */
  Wire.begin( );
/*  Start keypad object  */
  kpd0.begin( );
  Serial.print( "K0 start with pinState = " );
  Serial.println( kpd0.pinState_set( ), HEX );
  kpd1.begin( );
  Serial.print( "K1 start with pinState = " );
  Serial.println( kpd1.pinState_set( ), HEX );
  kpd2.begin( );
  Serial.print( "K2 start with pinState = " );
  Serial.println( kpd2.pinState_set( ), HEX );
  kpd3.begin( );
  Serial.print( "K3 start with pinState = " );
  Serial.println( kpd3.pinState_set( ), HEX );
}

void loop(){  
 char key0 = kpd0.getKey();
  if (key0){
    Serial.println(key0);
  }
 char key1 = kpd1.getKey();
 if (key1){
    Serial.println(key1);
  }
 char key2 = kpd2.getKey();
  if (key2){
    Serial.println(key2);
  }
  char key3 = kpd3.getKey();  
  if (key3){
    Serial.println(key3);
  }
}

joeyoung, 2020-08-25:
This code works just fine (although in the listing above, some formatting is lost and '*' gets left out after /)

I think your startup reporting of the initial state is perhaps indicating a problem with connections to the PCF8574s--the initial value should indeed only be FF. I think you get FFFF when the device at the port address is missing. Maybe the 4 devices are all set with the same address pin strapping, and you're getting I2C bus conflicts.

Instead of '1' as the last argument in the constructors you can use the literal PCF8574.

The 'jwire' parameter in the example is present to illustrate where the parameter goes in the argument list of the constructor when/if it is used, but it is shown in a way that works with an arduino the only has the Wire I2C bus. The purpose of this argument is to allow the libraries to function with a different I2C bus when they're used with processors that can have more than one I2C bus. When the parameter is omitted, the constructor defaults to Wire.

closed 2020-08-25 by joeyoung

Maes2ro, 2020-08-25:
Thanks Joe, much appreciated.

I'll hook the board back up later and do some more experiments.

Stirling