joeyoung/arduino_keypads #14: Keystrokes

fungreenfox, 2022-12-26:
With esp32, the keypad_i2c.h library performs repeating keystrokes.
Is there a function to stop repeating?

joeyoung, 2022-12-28:
There should not be any difference between arduino using esp32 and arduino with ATMEGA328 processors.

I suspect this issue results from a use problem. The Keypad library and all of the port-expander versions here requires that the access to the library, getKey() eg., be repeatedly called from loop() so that the port pin scanning and debounce timing functions properly. If, after calling the library to get a key, is followed by some code that prevents loop() functioning or otherwise does not repeatedly call the library, the value fetched will appear to be repeating. The library is not allowed to function properly.

fungreenfox, 2024-07-15:
Thank you. Went to another scripting that can do.

closed 2024-07-15 by fungreenfox