joeyoung/arduino_keypads #18: SPI Version

Alyxvance, 2024-12-08:
Hi, is it possible to maka a SPI version of this library for the MCP23S17?

joeyoung, 2024-12-10:
It should be possible, but I have not done one. 
You would lose the benefit of the I2C device addressability, and it would tie up the SPI bus as the Keypad library continuously scans the keys. These two defects could be eased somewhat if there were an external SPI multiplexing scheme that could allow some sharing of the SPI as long as the sharing didn't then hog the SPI.
Please share  what  compelling reason to use SPI would warrant the development, then I could get a couple MCP23S17s to work with.

Alyxvance, 2024-12-10:
The addressability is still given in the case of an MCP23Sxx. You use 4 lines, but with just one CS line you can control just as many expanders as with I2C because the SPI type is addressable like the I2C.

One advantage is that the SPI type is faster. 1Mhz vs 10Mhz or 8Mhz on the Arduino

Next, there is no pin 7 bug.

And the fact that it takes up the whole SPI protocol is not the problem in my opinion because the I2C bus is not used in any other way with your library.

Another suggestion here would be to work with the interrupt pins, but that seems to require a bit more programming.

Unfortunately, I don't have much knowledge of programming. Otherwise I would program it myself, or I'll try, but it's questionable whether it will work.

I would like to try using the SPI type, as I have a project that also includes rotary encoders. However, these are greatly slowed down by the I2C process. If I program the whole thing on the Mega2560 native, the rotary encoders work without any major problems. In conjunction with the MCP23017 or PCF8575, the whole thing is slowed down. The finished project then uses the Amege32u4. However, it does not have enough pins for 64 keys. I could also implement the whole thing with shift registers, but the hardware effort would then be somewhat greater.

joeyoung, 2024-12-11:
Speed is a valid point for large key arrays. I've found the I2C bus is maxed out with about 4 expanders (see HelloKpd64_MC17 example). So, I will order some of the MCP23Sxx devices and work up an expander lib for Keypad.

Although the library examples do not include other I2C devices, I have used this library with a keypad, temp sensor, rtc, eeprom, one-wire converter, etc. and all can quasi-simultaneously share the single I2C bus with the keypad. Except for very large key arrays, the Keypad does not hog the bus. I don't think that scenario works with SPI.

Alyxvance, 2024-12-12:
Well, that's the point I want to try out. Because the keypad works in and of itself. It just slows down the encoders that are connected directly to the Arduino. That's why I'm thinking of including the interrupts in the project.
But to do that you have to set a status in the expander so that it sets an interrupt to INTA or INTB when a key is pressed. When that's done, you can query the matrix.
At the moment I just interrupt the matrix query as soon as a pin on the rotary encoder sends a signal (goes low). That helps a bit but it's not the best solution, because if one of the rotary encoders is in half step, the matrix is ​​no longer queried.

If you have any tips for reprogramming or an example that might work, I'd try it out now and give you feedback, if you want.

and sorry for my bad englisch ;)

Alyxvance, 2024-12-13:
so I have made changes in my setup it works may be is it okay so but the interrupting shoud implementet in all versionen to use it optional. I am not sure how but fuer is fisnisch my project. 

here the changes i have made. 
CPP
```c++
/*
||
|| @file Keypad_MCS17.cpp
|| @version 2.0 SPI
|| @author G. D. (Joe) Young / SPI Alyx Vance
|| @contact "G. D. (Joe) Young" <jyoung@islandnet.com>
||
|| @description
|| | Keypad_MC17 provides an interface for using matrix keypads that
|| | are attached with Microchip MCP23017 I2C port expanders. It 
|| | supports multiple keypads, user selectable pins, and user
|| | defined keymaps.
|| | The MCP23017 is somewhat similar to the MCP23016 which is supported
|| | by the earlier library Keypad_MC16. The difference most useful for
|| | use with Keypad is the provision of internal pullup resistors on the
|| | pins used as inputs, eliminating the need to provide 16 external 
|| | resistors. The 23017 also has more comprehensive support for separate
|| | 8-bit ports instead of a single 16-bit port. However, this library
|| | assumes configuration as 16-bit port--IOCON.BANK = 0.
|| #
||
|| @version SPI 2.0 December 13, 2024
|| | Keypad_MCS17 is a SPI version of Keypad_MC17 from G. D. (Joe) Young
|| | SPI changes by Alyx Vance.
|| #  
||
|| @version 2.0 - April 5, 2020
|| | MKRZERO, ESP32 compile error from inheriting TwoWire that was OK with
|| | original ATMEGA boards; possibly because newer processors can have
|| | multiple I2C WireX ports. Consequently, added the ability to specify
|| | an alternate Wire as optional parameter in constructor.
|| #
||
|| @license
|| | This library is free software; you can redistribute it and/or
|| | modify it under the terms of the GNU Lesser General Public
|| | License as published by the Free Software Foundation; version
|| | 2.1 of the License or any later version.
|| |
|| | This library is distributed in the hope that it will be useful,
|| | but WITHOUT ANY WARRANTY; without even the implied warranty of
|| | MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
|| | Lesser General Public License for more details.
|| |
|| | You should have received a copy of the GNU Lesser General Public
|| | License along with this library; if not, see 
|| | <https://www.gnu.org/licenses/> 
|| #
||
*/

#include "Keypad_MCS17.h"

//  REGISTERS                   //  description                 datasheet
#define IODIRA          0x00    //  Data Direction Register A     P18
#define IODIRB          0x01    //  Data Direction Register B     P18
#define IOPOLA          0x02    //  Input Polarity A              P18
#define IOPOLB          0x03    //  Input Polarity B              P18
#define GPINTENA 		    0x04    //  Interrupt enable A            P19
#define GPINTENB    	  0x05    //  Interrupt enable B            P19
#define DEFVALA       	0x06    //  Interrupt default A           P19
#define DEFVALB       	0x07    //  Interrupt default B           P19
#define INTCONA         0x08    //  Interrupt control register A  P20
#define INTCONB       	0x09    //  Interrupt control register B  P20
#define IOCON          	0x0A    //  IO control register           P20
[... 254 lines cut ...]

private:
    byte cs_port;
    // I2C device address
    byte i2caddr;
	SPIClass *_spi;
	// I2C pin_write state persistant storage
	word pinState;
//	byte pin_iosetup( );
	// MCS17 setup
	word iodir_state;    // copy of IODIR register
	void _begin( void );
};

#endif // KEYPAD_MCS17_H

/*
|| @changelog
|| |
|| | 2.0 SPI 2024-12-13 - Alyx Vance : Made as SPI version for MCP23S17
|| | 2.0 2020-04-05 - Joe Young : MKRZERO compile error, Wire spec'd in Constructor
|| | 1.0 2014-05-18 - Joe Young : Convert from Keypad_MC16
|| #
*/
```

joeyoung, 2024-12-13:
I have not studied your modifications in detail, but it looks about right. When I receive the SPI parts, I will check it out, but since it's working for you, that's good enough.

Thanks for raising this issue.

Alyxvance, 2024-12-13:
Thanks. That's OK if you have the SPI models and you could test, than it would be nice if i hear you.

joeyoung, 2025-01-05:
Library Keypad_MCS17 is added to the repository.

Alyxvance, 2025-01-06:
Thank you. 
And a Happy New Year.

joeyoung, 2025-01-07:
Could be done. 

It's too bad the SPI interface library doesn't already have a function to set the bus speed like the Wire.setClock() call. But since the speed appears to be considered a device-characteristic limitation, perhaps the setting should be part of the device setup. So, I will try adding another function that would look like kpd.setClock( 20000000 ).

joeyoung, 2025-01-11:
Added the function kpdSettings( ) to permit changing the SPI bus characteristics which are normally specific to the specifications of the IC. The use of the function is optional as the settings are set by default (in begin( )) to the recommended values, but may be useful for testing purposes--say to reduce the clock speed of the bus. See wiki and example sketch for use.

closed 2025-01-11 by joeyoung